silvery107/rl-mpc-locomotion #1: Go1 Support

lucasbrodo, 2023-09-29:
Hi @silvery107,
Thanks for this very rich work.

Could you list some guidelines for deploying your code to Unitree Go1?

Thanks for the feedback.

Lucas

silvery107, 2023-09-30:
Since the MPC controller and the RL training environment are decoupled, you can deploy the trained policy network using `RobotRunnerPolicy`, and you will need to connect its interface to unitree sdk for sending motor commands and reading sensor data.

lucasbrodo, 2023-10-04:
Hi @silvery107,
I have connected the unitree SDK interface for sending motor torques commands and reading sensor data. I have only modified the `RL_MPC_Locomotion.py` and I'm just focusing on getting `RobotRunnerFSM` running for now.
I don't know exactly where to modify the script for providing the real robot foot forces/contacts since it's vital for the MPC.

Thank you very much for your help.

silvery107, 2023-10-08:
What do you mean by providing the foot forces/contacts? 

Please note that this codebase is not designed to run on a real robot (the sim2real part is not included), so run it with extreme caution and at your own risk.

lucasbrodo, 2023-10-09:
Sorry for the confusion, I thought that the foot contacts were used for the MPC controller but it appears that they are predicted (if I refer to the original MIT-Cheetah paper).

Could you indicate what is needed specifically for the deployment on the real robot ? Which parameters/values need to be modified ?

In my case, the controller outputs very high torques after providing the sensors' values from the real robot. The robot can't even stand up. 

Again, thanks for your help.

silvery107, 2023-10-09:
For parameter tuning, try modify the joint Kp, Kd and weights of MPC solver.

Also a filter (e.g. EKF) is needed for sensor data, since the current state estimator is basically a dummy version.

silvery107, 2023-10-18:
Closed due to inactivity

closed 2023-10-18 by silvery107