silvery107/rl-mpc-locomotion #2: Small shifting with zero command

linchangyi1, 2023-10-20:
Hi @silvery107,

Thanks a lot for releasing the nice project. During studying the code base, I met some issues and want to discuss them with you.

1. In the update function within /MPC_Controller/common/StateEstimator.py, the current code for updating vBody and omegaBody is as follows:
        self.result.vBody = self.result.rBody @ self.result.vWorld
        self.result.omegaBody = self.result.rBody @ self.result.omegaWorld
However, I believe the correct transformations should be:
        self.result.vBody = self.result.rBody.T @ self.result.vWorld
        self.result.omegaBody = self.result.rBody.T @ self.result.omegaWorld
This is based on the assumption that we want to transform the linear and angular velocities from the world frame to the body frame, where v_b = R_b_to_s @ v_s?

2. Additionally, I noticed a minor drift in the yaw-axis and the y-axis when the command is set to zero, as shown in the following video link:
https://github.com/silvery107/rl-mpc-locomotion/assets/71823391/0c03f991-0c5c-46b8-b499-c07a644657c9
Could you please give some suggestions on how to improve it?

Thanks a lot for your time and look forward to your response~~~

silvery107, 2023-10-21:
Hi Changyi,

Thanks for your interests in my project.

1. The transpose is done within `quat_to_rot()` due to the rotation convention from Cheetah Software.
2. The drift is because the MPC states I used are alined to the robot, i.e. no position constraints except height are enforced, so the drift is expected. You can use the world frame states and reference to eliminate that drift.

linchangyi1, 2023-10-21:
Got it~ Thanks a lot for your reply.

silvery107, 2023-10-21:
Resolved

closed 2023-10-21 by silvery107