silvery107/rl-mpc-locomotion #10: Hello, I want to study RL MPC.  Can you recommend some papers?

chendayuxixi, 2024-03-27:
thank you!!!!

silvery107, 2024-03-27:
Hi, thanks for your interests.

You can check the reference of my thesis attached. 

[References.pdf](https://github.com/silvery107/rl-mpc-locomotion/files/14780141/References.pdf)

chendayuxixi, 2024-03-29:
> Hi, thanks for your interests.
> 
> You can check the reference of my thesis attached.
> 
> [References.pdf](https://github.com/silvery107/rl-mpc-locomotion/files/14780141/References.pdf)

Is there a paper related to this open source code

zhouliu, 2024-07-31:
Hi, @silvery107! Your [slides](https://docs.google.com/presentation/d/18bznpYrkCPnhCisySPDz18hvL3Ytere7JiJEbdLvpgU/edit#slide=id.g1c504e3e94_2_127) are excellent! Could you give me a copy of your slides? Thank you!

silvery107, 2024-07-31:
Hi, thanks for your interests.
I can only share the pdf version of it.

[UG-Thesis-EN.pdf](https://github.com/user-attachments/files/16445652/UG-Thesis-EN.pdf)

wyr0130, 2025-05-19:
> 您好，感谢您的关注。
> 
> 您可以查看附件中我的论文的参考文献。
> 
> [参考文献.pdf](https://github.com/silvery107/rl-mpc-locomotion/files/14780141/References.pdf)

Hello. Is there a corresponding paper for this code? Can I have a look at your paper?

silvery107, 2025-05-19:
I can share my undergraduate thesis if you are really insterested, but I think the slides are more helpful for understanding the implementation.

wodewo365, 2025-05-20:
> I can share my undergraduate thesis if you are really insterested, but I think the slides are more helpful for understanding the implementation.

Hello, I feel that the paper is more helpful for understanding this project. If possible, please share the paper with me.

silvery107, 2025-05-21:
[庄宇伦-毕业论文.pdf](https://github.com/user-attachments/files/20359259/-.pdf)

wodewo365, 2025-05-21:
> [庄宇伦-毕业论文.pdf](https://github.com/user-attachments/files/20359259/-.pdf)

OK, thank you!