silvery107/rl-mpc-locomotion #13: How to change the step frequency?

zst1406217, 2024-04-13:
Hello! It is really a great work! When I run MPC controller, I find that the robot's step frequency is very high. I want to know how can I change it, thank you!

silvery107, 2024-04-23:
Thanks for your interests and sorry for my late response.

If you want to change the stepping frequency, try scale up the gait segments in [this file](https://github.com/silvery107/rl-mpc-locomotion/blob/19ef04815937f0c9cdac7ea57ed068477dfa8716/MPC_Controller/convex_MPC/ConvexMPCLocomotion.py#L30-L32), e.g.

```py
self.trotting = OffsetDurationGait(20, 
                    np.array([0, 5, 5, 0], dtype=DTYPE)*2, 
                    np.array([5, 5, 5, 5], dtype=DTYPE)*2, "Trotting")
```

zst1406217, 2024-04-25:
Thanks a lot for your reply!