silvery107/rl-mpc-locomotion #18: Are you willing to support  the  sim2real code  in  python ?   Thanks for reply.

heartInsert, 2024-08-01:
Hi, mate , I  have  deploy  my  RL  model  in   gazebo and real_robot   successfully  by  ROS .  But  when  I   run  it  in  gazebo   and  real  robot  ,  it   didn't  walk    as   smoothly as  in  isaacgym  .  The  whole posture   between  real  robot  and  isaacgym  are  very  closed.  But  the  position  will  changed  very  fast   in  real  robot ,  I'm not sure if this is related to the   controlFrequencyInv ，substeps  or  num_position_iterations  in  config ?   Did you   encounter  the  same  problem like me ?

silvery107, 2024-08-19:
Hi,
Could you provide a short video?
Try turn off the visualization in Isaacgym and measure the actual loop time in your script to see if that matches the desired control rate.
Thanks

silvery107, 2024-08-19:
Hi,

There could be a lot reasons, but sim2real support is out of the scope of this project.

Thanks and good luck with your project!

closed 2024-08-19 by silvery107

jiahe0000, 2025-06-09:
> Hi, mate , I have deploy my RL model in gazebo and real_robot successfully by ROS . But when I run it in gazebo and real robot , it didn't walk as smoothly as in isaacgym . The whole posture between real robot and isaacgym are very closed. But the position will changed very fast in real robot , I'm not sure if this is related to the controlFrequencyInv ，substeps or num_position_iterations in config ? Did you encounter the same problem like me ?
> 
> Thank you for your reply again.

Hello, I encountered some issues during the process of deploying the model to Gazebo for sim2sim verification. Could you share the part of the program deployed into Gazebo? Greatly appreciated!