silvery107/rl-mpc-locomotion #19: Legacy importing bugs

silvery107, 2024-09-09:
```
ImportError: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
NomoduleNotFoundError: No module named 'gym'
```

zhshubs, 2024-10-28:
I have met this bug many times, I solve this bug by copying the libpython3.8.so.1.0 to /user/lib. I guess that install the python3.8 in your system not the virtual enviroment can also solve it.

silvery107, 2024-10-28:
Hi @zhshubs, thanks for your comment. Do you think there is a way to solve this systematically? i.e. by changing the virtual environment setup (varying the version of python/packages)

If you find one, feel free to open a PR to this, thanks.

silvery107, 2025-02-07:
Quote from legged gym README.
> If you get the following error: `ImportError: libpython3.8m.so.1.0: cannot open shared object file: No such file or directory`, do: `sudo apt install libpython3.8`. It is also possible that you need to do `export LD_LIBRARY_PATH=/path/to/libpython/directory` / `export LD_LIBRARY_PATH=/path/to/conda/envs/your_env/lib` (for conda user. Replace /path/to/ to the corresponding path.).

closed 2025-02-07 by silvery107