silvery107/rl-mpc-locomotion #21: not

zhshubs, 2024-10-28:


silvery107, 2024-10-28:
You can check `gamepad_reader.py` and use the same library to read your keyboard inputs and map to corresponding commands.

Welcome to open a PR for that once you solve it!

zhshubs, 2024-10-30:
As you say, I check gamepad_reader.py and use the same lib to read my keyboard inputs, but it cant not map to corresponding commands. I am sure read my keyboard inputs successfully but can not control the robot in sim.

silvery107, 2024-10-30:
Could you be more specific by "it cant not map to corresponding commands"? What fields did you fill in with the keyboard inputs in`gamepad_reader.py` and what's the entry script of your "robot in sim" (like `RL_MPC_Locomotion.py`)?

silvery107, 2024-11-19:
Close due to inactivity.