silvery107/rl-mpc-locomotion #23: Try 'train.py' and 'ValueError: too many values to unpack (expected 2)' occured

MaRuiXiang1201, 2024-11-28:
Thank you for your work, this job is fabulous!
I tried the 'python train.py task=Aliengo headless=False' command and encountered a dimension mismatch error:
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Error executing job with overrides: ['task=Aliengo', 'headless=False']
Traceback (most recent call last):
  File "train.py", line 61, in launch_hydra
    ppo_runner = OnPolicyRunner(env, train_cfg_dict, log_dir, cfg.rl_device)
  File "/home/xxx/Downloads/rl-mpc-locomotion/extern/rsl_rl/rsl_rl/runners/on_policy_runner.py", line 30, in __init__
    obs, extras = self.env.get_observations()
ValueError: too many values to unpack (expected 2)

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
-------------------------------------------------------------------------------------------------------------------------------------------------------------
the 'self.env.get_observations()' is printed as follows:
self.env.get_observations(): 
tensor([[0., 0., 0.,  ..., 0., 0., 0.],
        [0., 0., 0.,  ..., 0., 0., 0.],
        [0., 0., 0.,  ..., 0., 0., 0.],
        ...,
        [0., 0., 0.,  ..., 0., 0., 0.],
        [0., 0., 0.,  ..., 0., 0., 0.],
        [0., 0., 0.,  ..., 0., 0., 0.]], device='cuda:0')
it seems to mismatch with the command 'obs, extras = self.env.get_observations()'
Some help will be appreciated~

silvery107, 2024-11-28:
Hi,

Thanks for your interests.

The submodule `rsl_rl` should be checked out to commit `2ad79cf`, instead of the latest commit. I am not sure if it's the issue of `git submodule update --init`. 

Could you try clone the repo again using `git clone https://github.com/silvery107/rl-mpc-locomotion.git --recurse`?

MaRuiXiang1201, 2024-11-29:
Thanks for your help!
It turns out to be the problem of `rsl_rl` version. The 2.0.2 version will rise the problem of `ValueError` and the 1.0.2 version works just fine.
When I first git clone using `git clone https://github.com/silvery107/rl-mpc-locomotion.git `and `git submodule update --init` sequentially, it won't be able to download all the submodules. Instead the command `git clone https://github.com/silvery107/rl-mpc-locomotion.git --recurse` works just fine.
Thank you again for you patience and help!

silvery107, 2024-11-29:
Great!

closed 2024-11-29 by silvery107