silvery107/rl-mpc-locomotion #25: Why the action output of the nn policy can update the qp_weights in MPC controller?

MaRuiXiang1201, 2024-12-05:
I have encountered two issues while trying to understand the logic of the code and would like to seek your help.
My understandings are listed as follows, please point mistakes out if any:

1. [Quadruoed.py](https://github.com/silvery107/rl-mpc-locomotion/blob/main/MPC_Controller/common/Quadruped.py) defines the `_mpc_weights`, which is the qp_weights in `mpc_osqp` for solving quadratic programming problems. The combination of RL and MPC is that the policy trained by RL algorithms can adjust the `_mpc_weights`.
2. Both Fsm and Min modes directly use the `_mpc_weights` defined in [Quadruoed.py](https://github.com/silvery107/rl-mpc-locomotion/blob/main/MPC_Controller/common/Quadruped.py) without any change, so they do not involve the combination of MPC and RL.
3. In Policy mode, [WeightPolicy.py](https://github.com/silvery107/rl-mpc-locomotion/blob/main/RL_Environment/WeightPolicy.py) loads a pretrained `.pt` file to set parameters for actor and critic networks(`self.policy = self.actor_critic.act_inference`), then puts observations into the policy to obtain `current_action`(`current_action` = self.policy(obs.detach())`).

Here are my questions:

1. Why the `current_action` can update the `_mpc_weights`? Is any reference literature or theory support?
`self.weights=actions_rescale.detach().cpu().numpy()[0]`
2. Why the `self.weights` changes only 3 times and no longer changes?
    When I run `python RL_MPC_Locomotion.py --robot=Aliengo --mode=Policy`, the `self.weights` changes as follows:
        self.weights: [ 1.   1.5  0.   0.   0.  50.   0.   0.   0.1  1.   1.   0.1]# initial weight
        self.weights: [ 1.  9.  9. 70. 70. 30.  2.  0.  2.  0.  0.  2.]             # first iteration
        self.weights: [ 1.  9.  1. 70. 70. 30.  2.  0.  2.  0.  0.  2.]             # second iteration
        self.weights: [ 9.  1.  1. 70. 70. 30.  0.  0.  0.  0.  2.  0.]             # third iteration and won't change anymore

silvery107, 2024-12-05:
Hi,

Thanks for your interets.

I think your understanding is correct.

Regrading your questions:
1. This is because I treat the RL training as a black box (zero-order) optimization of my hyperparameters conditioned by observations. You can refer to methods like bayesian optimization algorithm (BOA) and covariance matrix adaptation evolution strategy (CMA-ES).
2. It stop changing because the output already hits the boundary of the action space, which is defined in `Parameters.py'.

MaRuiXiang1201, 2024-12-06:
Got it, thanks for your reply~

closed 2024-12-06 by silvery107