silvery107/rl-mpc-locomotion #26: Aliengo action

zhagao380, 2024-12-12:
Hello, I would like to ask if a controller is required to execute the command ‘python RL_MPC_Locomotion.py --robot=Aliengo --mode=Policy --checkpoint=path/to/ckpt’. I tried using a controller to operate but it doesn’t respond.

silvery107, 2024-12-12:
Hi,

Could you verify if the output is correct by directly running `RL_Environment/gamepad_reader.py`?

zhagao380, 2024-12-13:
> 你好
> 
> 您能否通过直接运行来验证输出是否正确？`RL_Environment/gamepad_reader.py`

When I run the command python RL_MPC_Locomotion.py --robot=Aliengo, I can control the robot with the joystick, which indicates that the joystick output is fine. However, after adding --mode=Policy --checkpoint=path/to/ckpt, the joystick control no longer works. Therefore, I want to ask if adding these options means that the policy takes direct control and joystick control is no longer used?

zhagao380, 2024-12-13:
> Hi,
> 
> Could you verify if the output is correct by directly running `RL_Environment/gamepad_reader.py`?
I ran RL_Environment/gamepad_reader.py, and it outputted physical quantities such as speed and angular velocity.

silvery107, 2024-12-13:
Hi,

I don't think that option will disable the gamepad input, and if the robot can respond to LB and RB, it should be fine.

Maybe the policy is outputing zeros on the weights of velocity tracking which makes the robot trotting in place?

zhagao380, 2024-12-13:
Ok , thanks !

closed 2024-12-14 by silvery107