silvery107/rl-mpc-locomotion #27: isaacgym shows an error

ltf1001, 2024-12-19:
Thanks to the author for open-sourcing this algorithm. The idea and code structure are really perfect. But when I run it, an error occurs. When I follow your instructions:
python train.py task=Aliengo headless=False
to train the model, it reports an error and cannot enter the corresponding graphical interface. It shows no response and wants to exit. How should I solve this problem?

silvery107, 2024-12-19:
Could you paste the full error traceback here? I do suspect it's a driver side issue instead of the codebase.

ltf1001, 2024-12-20:
> Could you paste the full error traceback here? I do suspect it's a driver side issue instead of the codebase.

Thanks for your reply, this helped me solve the problem.

closed 2024-12-20 by ltf1001