silvery107/rl-mpc-locomotion #28: The version eigen3@02f4200 is no longer available

ling-1226, 2024-12-24:
The version eigen3@02f4200 is no longer available.Is there a specific reason why this commit must be used, or can other versions of Eigen be used instead?

silvery107, 2024-12-24:
Interesting, they just moved eigen3 from github to gitlab recently. 

And yes you should be able to compile with other versions of eigen3. If you did success, please feel free to report the version you chose.

ling-1226, 2024-12-24:
Thank you.I used version 3.3.7 of this link(https://github.com/eigenteam/eigen-git-mirror/tags) to successfully compile.Is there no gamepad to run this project?

silvery107, 2024-12-24:
Try append `--disable-gamepad --mode=Min` and edit `commands` in the entry script.

closed 2024-12-30 by silvery107