silvery107/rl-mpc-locomotion #30: Model replacement

zhagao380, 2025-04-02:
Hello, I would like to replace the robots in the project with our own robots, but after importing the model, there is a phenomenon of parts separation
As shown in the link:[image]

silvery107, 2025-04-03:
Change `flip_visual_attachments` to `True` if it was `False` or the other way around,  in `asset_options`.

zhagao380, 2025-04-07:
> 更改为 if it was or 相反，在 中。`flip_visual_attachments``True``False``asset_options`

Thank you for your previous answer. After making the changes you suggested, the problem of parts scattering was resolved, but the robot cannot stand up.

silvery107, 2025-04-07:
Could you describe your modifications and the process of how you start the program (with what launching options)?

zhagao380, 2025-04-08:
[image]
Modifications: asset_options.flip_visual_attachments = True.
Also, I don't know what you mean by launching options.

silvery107, 2025-04-08:
Please check the README for how to enter the controller with a gamepad or launch it with `--mode=Min`.

zhagao380, 2025-04-08:
I used to run python RL_MPC_Locomotion.py --robot=Aliengo this command , but now adding --mode=Min this command allows me to stand up. But I have two questions, one is why does Aliengo or A1 stand even without adding it; Secondly, I would like to train with the replaced robot in the future and use this command to execute it. Can you provide me with an idea for the operation?

silvery107, 2025-04-08:
The default launching option (`python RL_MPC_Locomotion.py`) runs the Finite State Machine which start with a very simple "recovery stand" controller. It could fail due to the different size of your custom robots.
Set `--mode=Min` will bypass the state machine and only runs the locomotion controller which is the convex MPC controller and it can handle more versitle motion and control commands.

And please follow the README closely with a gamepad~

closed 2025-04-10 by silvery107