silvery107/rl-mpc-locomotion #32: new model walking

lingengrou, 2025-04-16:
Hello, author. Thank you for sharing the project. After replacing the model with our own robot and running it in play mode, I found that it walks normally on flat ground, but when it encounters a slope, it instantly "springs" backward and rolls over. Could you please advise on how to resolve this issue?

silvery107, 2025-04-16:
Thanks for your interests.

Try check the estimation of `ground_normal`. You can serch the `flat_ground` flag and set it to `False`.

lingengrou, 2025-04-18:
Thank you for your response. I checked the `flat_ground` setting and found that it is set to `False` by default. I changed it to `True`, but the issue with the robot falling over on the slope still persists. Are there any other possible reasons that could be causing this problem?

silvery107, 2025-04-18:
Can you share a video of it?

lingengrou, 2025-04-19:
https://github.com/user-attachments/assets/b7f6322b-a226-47a7-8f3e-197f2b17a615

silvery107, 2025-04-19:
I never saw that happened before lol 
You can try using the tabs in Isaac viewer to check "show collision geometry" and verify where is the actual collision boundary. Maybe it's because the collision mesh has a very different shape than sphere or box?

lingengrou, 2025-04-22:
[image]
Thanks for your reply. I enabled "render collision meshes" to inspect the robot's collision models, and I noticed that the displayed shapes are irregular and closely conform to the visual mesh. In contrast, the collision models used in the project's AlienGo are regular geometric shapes like boxes and spheres. Could this difference be the reason why the robot bounces or gets launched when walking up a slope?

silvery107, 2025-04-22:
Usually that is fine. Complex shapes could make computation slower. Maybe you can check what happens when the robot walk on to the slope with collision rendering on, and zoom in to see if the mesh has any unclosed shapes at contacts or any penetrations?

lingengrou, 2025-04-28:
Thank you！ I have already solved this issue — I replaced the foot collision models with spheres.

closed 2025-04-28 by silvery107