silvery107/rl-mpc-locomotion #43: Problems with handle control.

shabahhhshsbha, 2025-09-09:
Hello, I've looked at your rl-mpc-locomotion project and tried running the code myself, but I've been having issues with the controls. When I use the controller to make the dog move forward, backward, left, or right, the dog doesn't respond. It keeps moving in reverse or other states and can't be controlled. Could you tell me what the problem might be?

silvery107, 2025-09-09:
Sounds like a controller issue. Could you try directly input the desired commands in this [line](https://github.com/silvery107/rl-mpc-locomotion/blob/0d7bd9a7e197eebebe8e5bf79b1740608907b6a2/RL_MPC_Locomotion.py#L87)?

shabahhhshsbha, 2025-09-10:
It's okay, but the commands entered in this way are some fixed commands and not flexible enough.